xil-se/xildebug_sw #19: usb: Bringup basic CDC and HID endpoints

kbeckmann, 2018-04-21:


arturo182, 2018-04-22:
Phew, just an issue, I thought you already did that too :P

kbeckmann, 2018-04-22:
I think we need to think a bit how we want this to work. My suggestion:

1 bi-directional CDC should be 100% connected to the UART - no bullshit. Should change the UART baud rate etc based on the CDC commands.
1 HID endpoint for CMSIS-DAP. Have no clue how this works but you make it sound so easy Artur :peanuts: 
1 bi-directional endpoint for ADC and control. Could be CDC or HID. Benefit with CDC is that you can just use something like minicom to interact with it.

kbeckmann, 2018-04-26:
CDC and HID endpoints implemented now, next up is being able to have them both activated simultaneously. GLHF!

arturo182, 2018-04-26:
Yeah done in #30 and #32 :D

closed 2018-05-06 by arturo182