xil-se/xildebug_sw #2: Test value

arturo182, 2018-04-21:
https://github.com/kbeckmann/xildebug/blob/fe7ccd1fc2e7b4f892d373cbefc386ada887a9ee/app/drivers/mcp4018t.c#L6

What voltage does the test value set and the default value as well, if we have a  1.8V device connected and we restart xildebug, what if it sets the voltage to 5V or something. Would be bad.

Also regarding the API, maybe instead of set_value we should have set_voltage, easier to understand.

kbeckmann, 2018-04-21:
> Also regarding the API, maybe instead of set_value we should have set_voltage, easier to understand.

My idea here was that the driver for the chip should only handle the raw communication. Voltage etc will be handled in a separate module that also will toggle the DUT_VDD_EN pin

arturo182, 2018-04-21:
Ok, many layers, that works.

kbeckmann, 2018-04-21:
> What voltage does the test value set and the default value as well, if we have a 1.8V device connected and we restart xildebug, what if it sets the voltage to 5V or something. Would be bad.

Yeah I thought about that as well. We have to make sure that DUT_VDD_EN is low when the test is executed, that way the DUT shouldn't feel a thing (tm).

arturo182, 2018-04-21:
( ͡° ͜ʖ ͡°)

kbeckmann, 2018-04-26:
Fixed now (not enabling the VDD_DUT_EN before tests)

closed 2018-04-26 by kbeckmann