xil-se/xildebug_sw #5: Fix to use proper p_name at some point

arturo182, 2018-04-21:
https://github.com/xil-se/xildebug_sw/blob/b5ef8b66cc7f98fa468062f20203c5f07501058a/app/drivers/adc.c#L51

https://github.com/xil-se/xildebug_sw/blob/b5ef8b66cc7f98fa468062f20203c5f07501058a/app/drivers/adc.c#L65

kbeckmann, 2018-04-21:
You mean change the argument names to p_adc_handle or something? I think we might have a lot of these in the whole codebase.

arturo182, 2018-04-21:
Yeah, exactly what I meant, and move the asterisk to the right, need to follow them kernel style rules ;)

kbeckmann, 2018-04-21:
To be honest, the functions that have this issue are functions that are declared in the SDK. Changing argument names feels a bit so-so, they should ideally match with the .h file.

closed 2018-04-26 by kbeckmann