xil-se/xildebug_sw #55: Have a consistent getter/setter pattern

kbeckmann, 2018-04-30:
I messed up here a bit, sometimes i use something_set, other times it's set_something.

What should we use? something_set feels a bit more api-ish than set_something, right? @arturo182

arturo182, 2018-04-30:
I think it should be `modulename_set_thing`

kbeckmann, 2018-04-30:
err_t power_set _dut_ldo
Why space 😪

kbeckmann, 2018-04-30:
Fatet discussing with frazz and a few 🍻later, we think that power_dut_enabled_get makes the lost sense. 
Reason is that if you sort the calls it all makes sense. Module > narrower scope > set/get. Get and set should be close to each other.

arturo182, 2018-04-30:
I don't agree, it signs like Yoda speak, "power duty enabled get, hihihi" https://youtu.be/5I671B4MCC0

kbeckmann, 2018-05-01:
@EnJens we need your input 🤙

arturo182, 2018-05-01:
So we should have 
```
adc_callback_set
max14662_value_set
max14662_bit_set
max14662_value_cached_get
max14662_value_get
mcp4018t_value_set
mcp4018t_value_get
uart_handle_get
``` 
?

That all sounds so wrong :/

kbeckmann, 2018-05-01:
https://github.com/mossmann/hackrf/blob/master/firmware/common/max2837.h

They have some inconsistency there as well.. max2837_set_frequency vs max2837_reg_read/max2837_reg_write. Seems people mix these things up all the time.

arturo182, 2018-05-09:
I think this is the highest prio issue right now ;)