xil-se/xildebug_sw #63: Proper error handler

kbeckmann, 2018-05-05:
Not sure how much energy we should spend on this but it kinda sucks to do
```
if (r != ERR_OK)
	while (1) ;
```

We don't need to do fancy ram dumping and so on, but just halt the cpu and perhaps disable interrupts and that we get a central point where we can set a breakpoint when debugging.

arturo182, 2018-05-05:
It was just made like this because when I wrote it we had no UART support, now we could actually print to uart or do something else that's smart :P

kbeckmann, 2018-05-05:
Ideally we should print it to the control cdc that hasn't been implemented yet.. :P

arturo182, 2018-05-09:
If we implement the HID CLI mentioned in #44 then we could actually have ram dumping and then fetching over CLI. But we would need to use 64K of flash for the dump area.

kbeckmann, 2018-05-09:
Think we can survive with an error handler that stalls the cpu and lets you attach with a debugger..

arturo182, 2018-05-09:
Debugging a debugger :O